curtacircuitos/pcb-tools #8: Merge svgwrite and cairo backends

phsilva, 2015-02-05:
I will work next few days to merge current svg and cairo backends to get only one cairo based backend working with everything we already support.

As cairo supports various surface types, including SVG, I think we could remove code and end up with better render support and quality over different output types.

Let me know if you have any ideas or suggestions.

hamiltonkibbe, 2015-02-05:
Just make sure that getting cairo set up on windows is not too difficult before dropping svgwrite. Having cairo as an optional dependency would be idea. It could be discouraging to people trying to use this library only for transforms etc. if they had to struggle with getting cairo and pycairo set up, for features they wouldn't use.

I think a good goal would be a generic context class that uses an appropriate backend based on the file extension of the output filename (or at least a file type argument) and the currently installed backend libraries, e.g. doing something like (pseudocode):

    context = generic_context()
    some_gerber.render_to_image('silk.svg', context)

Would use either SVGwrite or Cairo depending on what the user has installed, and raise an exception if they didn't have any installed libraries that supported writing SVGs.

As far as consolidating code goes, I tried to put a bunch of convenience methods/properties into the primitives classes. Stuff like position, lower_left, etc. That way you can use a rendering library that draws rectangles with the origin in a corner or one that draws rectangles with the origin in the center without repeating all the stuff like `position.x - (rectangle.width / 2.)` in the rendering module to translate from center-referenced to corner-referenced rectangles.

phsilva, 2015-09-10:
#40 to solve this.

phsilva, 2015-09-10:
closed by #40

closed 2015-09-10 by phsilva