curtacircuitos/pcb-tools #12: Stroke with rectangular aperture

phsilva, 2015-02-11:
Current code does not pass enough information to be able to stroke rectangular apertures.

By spec, Gerber allows solid circle and solid rectangle apertures. Currenly we assume we are dealing with circular apertures and do not pass enough info to Line primitive to be able to render it. See attached example (which currently causes an exception for us).

```
%MOMM*%
%FSLAX24Y24*%
%ADD10R,2.0000X3.0000*%
D10*
X000000Y000000D02*
X100000Y100000D01*
M02*
```

which should render to (using gerbv):

[image]

phsilva, 2015-02-13:
Arcs cannot be stroked with anything but solid circle aperture. So I think we are good now.

closed 2015-02-13 by phsilva

pklaus, 2015-02-20:
Currently, rendering Arcs may fail with:

    File "../python3.4/site-packages/gerber/__main__.py", line 38, in <module>
        gerberfile.render(ctx)
      File ".../python3.4/site-packages/gerber/cam.py", line 246, in render
        ctx.render(p)
       File ".../python3.4/site-packages/gerber/render/render.py", line 131, in render
        self._render_arc(primitive, color)
      File ".../python3.4/site-packages/gerber/render/svgwrite_backend.py", line 84, in _render_arc
        width = arc.width if arc.width != 0 else 0.001
    AttributeError: 'Arc' object has no attribute 'width'

Since 8f69c1d Arc objects don't have a .width attribute anymore.