curtacircuitos/pcb-tools #25: to_metric / to_inch is not idempotent

punkkeks, 2015-04-24:
```
In [58]: b.to_metric()

In [59]: b.lower_left
Out[59]: (998.89, 1904.43)

In [60]: b.to_metric()

In [61]: b.lower_left
Out[61]: (25386.446, 48386.43)

In [62]: b.to_metric()

In [63]: b.lower_left
Out[63]: (644830.3683999999, 1229029.23)

In [64]: b.to_metric()

In [65]: b.lower_left
Out[65]: (16378705.997359997, 31217356.349999998)
```

This is a problem. A fix would be to store internally what unit is currently in use.

phsilva, 2015-04-24:
@punkkeks, I will take a look at it. Thanks for the catch.

phsilva, 2015-04-24:
Fixed by #25. Thanks.

closed 2015-04-24 by phsilva

punkkeks, 2015-04-24:
@phsilva please look at my comment 
https://github.com/curtacircuitos/pcb-tools/commit/390838fc8b70c9b105fdc1d3e35a4533b27faa83#commitcomment-10892665

punkkeks, 2015-04-24:
I would suggest, that you store all variables that need to be converted in a list as 'private member' (or similar) and convert these in the to_metric method in the primitive class... This could be easier and reduce the code a lot!

phsilva, 2015-04-24:
yep, it will reduce much of the code. will add this to my (growing) refactoring list. tks.

hamiltonkibbe, 2015-04-24:
I like that approach, I'll go ahead and make that change, along with making sure all the tests cover regression on this issue