curtacircuitos/pcb-tools #30: drill file path optimizer: drl-optimize.py

koppi, 2015-06-08:
I wrote a drill file path optimizer.

example:

input
[image]
optimized output:
[image]
code:
```python
#!/usr/bin/env python
#
# drl file TSP path optimizer
#

import sys
import math
import gerber
from tsp_solver.greedy import solve_tsp

# https://github.com/dmishin/tsp-solver
# https://github.com/curtacircuitos/pcb-tools

if len(sys.argv) != 2:
    print "usage: drl-optimize.py [drill-file.drl]"
    sys.exit(0)

f = gerber.read(sys.argv[1])
#print(f.report())

cnt = {}
l   = {}
t   = {}

for a,b in f.hits:
    n = a.number
    if n not in cnt.keys():
        cnt[n] = 0
        l[n]   = []
        t[n]   = (a,b)
    cnt[n] += 1
    l[n].append(b)

m = {}

for key, value in cnt.iteritems():
#    print key, value
    m[key] = [[0 for x in range(value)] for x in range(value)]

hits = {}

for key, value in cnt.iteritems():
    for i in range(0,value):
        for j in range(0,value):
            m[key][i][j] = math.hypot(l[key][i][0] - l[key][j][0], l[key][i][1] - l[key][j][1])

    path = solve_tsp( m[key] )
#    print path

    hits[key] = []

    for p in path:
#        print l[key][p]
        hits[key].append((t[key][0], l[key][p]))

print """
M48
FMAT,2
METRIC,TZ
"""

for i in t.keys():
    print "T%dC%f" % (t[i][0].number, t[i][0].diameter)

print """
%
G90
G05
M71
"""

for t in t.keys():
    print "T%d" % (t)
    for p in hits[t]:
        print "X%fY%f" % (p[1][0], p[1][1])

print """
T0
M30
"""
```

My code sucks, because I could not find any way to re-order the drl file statements in gerber.py.

phsilva, 2015-06-09:
Good to see someone using this code, even tough it is still lacking some good API for writing as in your case. I will see how we can improve this in the couple of days, ideas are much welcome.

phsilva, 2015-06-09:
@koppi, btw, what software is this one form your screenshots?

koppi, 2015-06-09:
@phsilva http://openscam.com/

hamiltonkibbe, 2015-06-09:
I'd been thinking a little bit about a way to link primitives to their respective statements so that you could make bi-directional changes, e.g. change a statement and have the primitives update automatically and vice-versa. One hack-ish option that wouldn't require any sort of primitive <--> statement link would be to iterate over all the statements in a file and regenerate all the primitives  when a statement is changed, but it doesn't really work the other way around. 

Then there are some gotchas, like the repeathole statement that would have to link with multiple drill hole primitives... Maybe break all of those statements down into individual coordinate statements when loading a drill file and then have some sort of "file optimization" step before exporting that groups drill hits and generates repeathole statements from the individual steps where appropriate.

hamiltonkibbe, 2015-06-25:
@koppi I've made some changes (on the statement_link branch) that lets you re-order drill hits in the file. If you change the order of the drill hits in `f.hits` in your example, when you call `f.write()` to export the gerber file, the drill hits will be in the new order. Check it out when you have a chance

koppi, 2015-06-25:
@hamiltonkibbe thank's for the update. pcb2gcode implemented the TSP optimization two weeks ago, see https://github.com/pcb2gcode/pcb2gcode/issues/12 I'm no longer interested in my hackish Python script, if someone else want's to give it a try, please do so.

hamiltonkibbe, 2015-07-23:
#31 Added support for re-ordering drill hits to allow operations like this, as well as an example based on this script.

closed 2015-07-23 by hamiltonkibbe