curtacircuitos/pcb-tools #32: holes in inner copper layer

CreativeSoftware, 2015-07-22:
When it has to handle internal holes in inner copper layers , pcb-tools uses lines instead of arcs. So the geometry becomes odd. Probably it is a problem with a gerber keyword

hamiltonkibbe, 2015-07-22:
Can you give an example file that displays this behavior? 

Some CAD programs (e.g. Altium) use gerber Region statements to define copper layers. Because region statements don't support arcs, circles are approximated by a series of straight lines. In altium there is a setting to configure how many lines to use when approximating circles, not sure about other cad programs. Have you compared the results with outputs from another gerber viewer?

CreativeSoftware, 2015-07-23:
I can give you a simple example that illustrates what I'm saying :)
Incorret rendering with pcb-tools:

https://dl.dropboxusercontent.com/u/710615/Bug.PNG

Correct rendering with gerbv :
https://dl.dropboxusercontent.com/u/710615/CorrectRenderGerbv.PNG

hamiltonkibbe, 2015-07-23:
Could you provide the Gerber file for that layer so I can try to figure out what's going on?
Just for reference, what CAD program was used to generate the file?

CreativeSoftware, 2015-07-23:
At this moment I don't know the name.
I cannot provide the full gerber file( proprietary issue), however I think that you are able to use this one:

https://dl.dropboxusercontent.com/u/710615/Desktop.7z

hamiltonkibbe, 2015-07-24:
Neither of those layers seem to have circular plane cutouts, could you just provide the layer that was rendering incorrectly?

CreativeSoftware, 2015-07-24:
https://dl.dropboxusercontent.com/u/710615/LA1.gbx
i.e.

hamiltonkibbe, 2015-07-24:
Thanks I'll take a look at that tonight

CreativeSoftware, 2015-07-29:
@hamiltonkibbe any news? I think that I can help if you need, I don't know how :)

hamiltonkibbe, 2015-08-03:
I started taking a look at it over the weekend, I'll let you know when I get it figured out.

hamiltonkibbe, 2015-08-09:
Think I got it sorted out:

[image]

CreativeSoftware, 2015-08-10:
Wel done !  It is correct ! :)

closed 2015-08-10 by hamiltonkibbe

CreativeSoftware, 2015-09-07:
Could you post the sample code that generates this image?
I'm not able to reproduce the same image. It could be a code problem or I've installed different libs or branches.
(currently I have the branch from 9/08)

hamiltonkibbe, 2015-09-13:
Can you try with the current version of master? A bunch of changes got merged in that affect rendering.  The code to generate that image would be nearly identical to the cairo_examply.py script in the examples folder. Something like this should work:

    from gerber import read
    from gerber.render import GerberCairoContext
    ctx = GerberCairoContext()
    layer = read('LA1.gbx')
    layer.render(ctx)
    ctx.dump('LA1.png')