curtacircuitos/pcb-tools #33: Svgwrite and cairo merge issues

CreativeSoftware, 2015-07-22:
After the merge I faced some issues. When I try load 2 or more layers on the same ctx dumping in the end to a png it only appears the last load, the first ones didn’t load.
When I try to use svg as output I have a problem probably because I’m using windows. It is prompted a problem on a temporary file and the svg it is not created.

phsilva, 2015-07-24:
@CreativeSoftware, could you try the latest version and see if it is solved? Just this particular issue for now.

CreativeSoftware, 2015-07-24:
The layers are been rendering properly overlaid ! Well done.

However, it is not working the svg.
 File "build\bdist.win-amd64\egg\gerber\cam.py", line 264, in render
  File "build\bdist.win-amd64\egg\gerber\render\cairo_backend.py", line 157, in dump
IOError: [Errno 13] Permission denied: 'c:\\users\\marcoe~1\\appdata\\local\\temp\\tmpx0icyj'

hamiltonkibbe, 2015-07-25:
@CreativeSoftware could you confirm that the permissions error is fixed with the last commit?

CreativeSoftware, 2015-07-27:
Svg writing permissions solved!

hamiltonkibbe, 2015-07-27:
@phsilva we should be able to close this, as it is specific to the cairo-render-unification branch, but I'm not sure if you want to wait until the branch is merged into master

phsilva, 2015-07-27:
@hamiltonkibbe, we can close, I will finish the work on the cairo-render-unification to get it ready to merge.

hamiltonkibbe, 2015-07-27:
Sounds good. Closing

closed 2015-07-27 by hamiltonkibbe