curtacircuitos/pcb-tools #37: Do not read solely from files

punkkeks, 2015-09-03:
When parsing a (e.g.) gerber file, you have to have a file, but in my current situation I already read the file because of internal file handling.
I'd like to have a function where you can pass the data and get the result back (I hope you undestand what I want, i can't express myself today :/ )

hamiltonkibbe, 2015-09-03:
Would the data be in a string or some other format (open file handle, etc.)?

punkkeks, 2015-09-03:
in my case a bytearray, but a string would be perfectly fine as well.

hamiltonkibbe, 2015-10-10:
@punkkeks Could you take a look at the read_from_memory branch and see if that works for your use case?

punkkeks, 2015-10-12:
Looks good to me. I tried to load the file (why do you call it loads?), get the primitives out and render an image. Works :) No Exceptions for me, so I give it a :+1:

hamiltonkibbe, 2015-10-12:
Probably too much messing around with the JSON module which uses load() for files and loads() for strings

phsilva, 2015-10-14:
fixed by #41

phsilva, 2015-10-14:
let me know if there is anything else on this issue, otherwise I'll close it.

closed 2015-11-18 by hamiltonkibbe