curtacircuitos/pcb-tools #38: Let the render write into memory

punkkeks, 2015-09-03:
When I want to draw an image for e.g. a webapplication, I don't want to save that image to the disk, but instead I want to send it via HTTP GET 'answer' to the client. Please provide me a method, so the render method will give me a bytearray back.

punkkeks, 2015-09-03:
Got a workaround, but that isn't nice

    from StringIO import StringIO

    sio = StringIO()
    ctx.dwg.write(sio)

hamiltonkibbe, 2015-12-21:
This should be fixed, I've added a ```dump_str()``` method to ```GerberCairoContext``` that returns a png as a string. Theres also a ```dump_svg_str()``` method which returns an SVG as a string.
Let me know if this works for your use case