curtacircuitos/pcb-tools #39: wrong svg render

punkkeks, 2015-09-08:
I have a gerber outline file with some lines and two arcs. Those two arcs are render in the wrong direction (they should be convex). 

[image]

    G04 Layer_Color=16711935*
    %FSLAX44Y44*%
    %MOMM*%
    G71*
    G01*
    G75*
    %ADD20C,0.1000*%
    D20*
    X383000Y64000D02*
    G03*
    X383000Y74000I97J5000D01*
    G01*
    X313000Y74000D02*
    G03*
    X313000Y64000I-97J-5000D01*
    G01*
    Y74000D02*
    X383000Y74000D01*
    X313000Y64000D02*
    X383000D01*
    X692500Y0D02*
    Y611000D01*
    X0D02*
    X692500D01*
    X0Y0D02*
    X692500D01*
    X0D02*
    Y611000D01*
    M02*

phsilva, 2015-09-10:
@punkkeks, are you using master or cairo-render-unification branch (soon to be merge back to master)?

punkkeks, 2015-09-10:
master

10. September 2015 13:37 Uhr, "Paulo Henrique Silva"  schrieb:
	@punkkeks (https://github.com/punkkeks), are you using master or cairo-render-unification branch (soon to be merge back to master)?

phsilva, 2015-09-10:
Ok, can you try cairo-unification-branch please? We are moving all the rendering to a single backend and hopefully we can fix it there if the problem is still present.

punkkeks, 2015-09-10:
I'm now on the "cairo-render-unification"-branch:

first, the svgcontext:

> In [11]: a = gerber.read('board.GM1')
>
> In [12]: ctx = gerber.render.GerberSvgContext()
>
> In [13]: a.render(ctx, 'a.svg')
> ---------------------------------------------------------------------------
> AttributeError                            Traceback (most recent call
> last)
> <ipython-input-13-f4efde6c2d4d> in <module>()
> ----> 1 a.render(ctx, 'a.svg')
>
> /home/marcel/build/pcb-tools/gerber/cam.pyc in render(self, ctx, filename)
>     255         """
>     256         ctx.set_bounds(self.bounds)
> --> 257         ctx._paint_background()
>     258         if ctx.invert:
>     259             ctx._paint_inverted_layer()
>
> AttributeError: 'GerberSvgContext' object has no attribute
> '_paint_background'

the cairocontext does it right, but the file is 1.2MB big....

$ file a.png
a.png: PNG image data, 20775 x 18330, 8-bit/color RGB, non-interlaced

I provided you the gerber file in the first post, so feel free to do it
yourself ;)

punkkeks, 2015-10-12:
Any progress?

phsilva, 2015-10-14:
@punkkeks, not yet, will get more time into this this week, will let you know.

brouhaha, 2015-11-15:
I'm trying to package pcb-tools for Fedora, using the latest commit (ebc8518), and I'm also seeing the "AttributeError: 'GerberSvgContext' object has no attribute '_paint_background'" error, when I try the example in the README.

closed 2022-06-02 by hellow554