curtacircuitos/pcb-tools #44: no attribute '_paint_background'

brouhaha, 2015-11-15:
I'm trying to package pcb-tools for Fedora, using the latest commit (ebc8518), and I'm seeing the "AttributeError: 'GerberSvgContext' object has no attribute '_paint_background'" problem which punkkeks reported in issue #39, when I try the example in the README.

I kludged a workaround for the "no attribute 'paint_background'" problem by patching svgwrite_backend.py to add a no-op _paint_background() method.

```
diff -up curtacircuitos-pcb-tools-ebc8518/gerber/render/svgwrite_backend.py.paint-background curtacircuitos-pcb-tools-ebc8518/gerber/render/svgwrite_backend.py
--- curtacircuitos-pcb-tools-ebc8518/gerber/render/svgwrite_backend.py.paint-background	2015-11-15 00:01:28.923538499 -0700
+++ curtacircuitos-pcb-tools-ebc8518/gerber/render/svgwrite_backend.py	2015-11-15 00:02:01.471283949 -0700
@@ -128,3 +128,6 @@ class GerberSvgContext(GerberContext):
         hit = self.dwg.circle(center=center, r=SCALE * circle.radius,
                               fill=svg_color(color))
         self.dwg.add(hit)
+
+    def _paint_background(self):
+        pass
```

phsilva, 2015-11-15:
Our fault here, the svgwrite backend is deprecated and broken. You can se thr Cairo backend to write both PNG and svg.

I will remove it to avoid confusion.

phsilva, 2015-11-16:
#45 will be merged as soon as @hamiltonkibbe agrees. Sorry for the confusion on this.

hamiltonkibbe, 2015-11-18:
Hey @brouhaha, could you give it a try now? This should have been fixed by ba82d60cb8a886a82d4c4199473bb1189efc6419.

hamiltonkibbe, 2015-12-20:
I'm gonna say we should go ahead and close this one, It should have been fixed by removal of SVGWrite backend

phsilva, 2015-12-21:
Agreed, #45 fixed this, just left open waiting for confirmation...

closed 2015-12-21 by phsilva