curtacircuitos/pcb-tools #50: Semitransparent layer

sergeyfedotov, 2015-12-12:
Is it possible to draw a semitransparent solder mask over the copper layer?

Like this:
[image]

hamiltonkibbe, 2015-12-14:
You can set the ```alpha``` parameter on the rendering context to set layer opacity, but at the moment we don't have the ability to render the soldermask layer in negative....

I'll take a look at this, shouldn't be too difficult to add

phsilva, 2015-12-21:
Fixed per #50. Would be nice if you could check and maybe post an image here from your use case. Thanks.

closed 2015-12-21 by phsilva

sergeyfedotov, 2015-12-21:
Thx, it works fine

[image]