curtacircuitos/pcb-tools #56: how can I read tessel file

ghotiv, 2016-09-11:
dear,
[Uploading tessel-v2-4-28-2015.zip…]()
this is my file, i read this unzip file use the code
`import os
from gerber import PCB
from gerber.render import GerberCairoContext, theme

GERBER_FOLDER = os.path.abspath(os.path.join(os.path.dirname(__file__), 'tessel-v2-4-28-2015'))

# Create a new drawing context
ctx = GerberCairoContext()

# Create a new PCB instance
pcb = PCB.from_directory(GERBER_FOLDER)
print pcb.layer_count
print pcb.board_bounds

# Render PCB top view
ctx.render_layers(pcb.top_layers,
                  os.path.join(os.path.dirname(__file__), 'pcb_top.png',),
                  theme.THEMES['OSH Park'])

# Render PCB bottom view
ctx.render_layers(pcb.bottom_layers,
                  os.path.join(os.path.dirname(__file__), 'pcb_bottom.png'),
                  theme.THEMES['OSH Park'])

# Render copper layers only
ctx.render_layers(pcb.copper_layers + pcb.drill_layers,
                  os.path.join(os.path.dirname(__file__),
                               'pcb_transparent_copper.png'),
                  theme.THEMES['Transparent Copper'])`

result like this
`Traceback (most recent call last):
  File "/home/ghoti/project_one/pcb_test/pcb_example.py", line 39, in <module>
    ctx.render_layers(pcb.top_layers,
  File "/home/ghoti/project_one/pcb_test/gerber/pcb.py", line 74, in top_layers
    return [board_layers[0]] + drill_layers + board_layers[1:]
IndexError: list index out of range`

hamiltonkibbe, 2016-09-13:
@ghotiv can you re-upload the zip file you were trying to open? the link seems to go back to this issue and not the gerber files

hamiltonkibbe, 2016-09-29:
@ghotiv, It seems the issue was with it not detecting layer types, I've added some more hints to the layer type detection, should be fixed as of fedaedb6ebb8cc6abfc218d224a3ab69bb71b56.
Let me know if it works for you