curtacircuitos/pcb-tools #60: .png resolution

ju5t, 2016-11-21:
What would be the best way to make sure a png file is generated with a fixed max-width? I tried setting up an example with a set of gerber files I can't share, but it generates an image of 26066 x 11066 pixels. Aside from it taking >5 minutes it's a little over the top right in this case.

Have I missed an option somewhere?

hamiltonkibbe, 2016-11-21:
`GerberCairoContext()` takes a scale parameter, which is the DPI of the output file. It defaults to 300, you can lower that to make the output image smaller. 

That's the best you can do for now, although I'm going to keep this open, since specifying a desired output size would be a useful feature.

ju5t, 2016-11-28:
Sounds good. Is there anything else I can do to speed things up? If I compare the speed to the gerbv cli it is outperformed by miles. Gerbv isn't as pretty though.

hamiltonkibbe, 2016-12-13:
7c4ec8a768c7dfd3124 speeds up rendering quite a bit.  @ju5t can you try with the current master branch, and see if that improves things for you?

ju5t, 2016-12-13:
This definitely seems to be much, much faster. I still have to test it with more complex gerber files but it definitely looks good. A 300 dpi image renders in <30 seconds now. Speed wise I think this sorts it. 

A custom max width or height for example would still be useful as it seems that changing the dpi is unreliable at best. A 15 dpi image can either be a reasonable size or so small that you can't tell the difference.

ju5t, 2016-12-13:
@hamiltonkibbe this is the speed difference on a 12 layer:

On the previous version: succeeded in 1252.20317434s
On the current master branch: succeeded in 58.22679924s

hamiltonkibbe, 2016-12-14:
Thanks for the benchmark @ju5t. I think there's still some optimizations to do there...

As far as the sizing goes, check out the `auto_scale_renders` branch. adds `max_width` and `max_height` arguments to `render_layers` on the cairo renderer.  If that works against a few more test cases than the ones I've done, I'll get that in master

ju5t, 2016-12-14:
Looks good @hamiltonkibbe.

ju5t, 2017-02-21:
@hamiltonkibbe just so you know I haven't seen any issues with the `auto_scale_renders` branch.

hamiltonkibbe, 2017-02-24:
Thanks @ju5t. Changes are in master

closed 2017-02-24 by hamiltonkibbe