curtacircuitos/pcb-tools #64: Calculating edges

ju5t, 2016-12-14:
There are cases where it miscalculates the edges of the pcb. This happened with the master branch too so wasn't introduced by `auto_scale_renders`. I haven't checked the files with anything else than gerbv yet though. Have you seen this before?

This is with gerbv:
https://www.dropbox.com/s/s64gng4yqf3d82n/gerbv-output.png?dl=0

This is with pcb-tools
https://www.dropbox.com/s/gbcvhp3lr68grd9/pcb-tools-output.png?dl=0

ju5t, 2017-10-08:
I'll close this for now as apparently I've removed the files from Dropbox.

ju5t, 2017-11-07:
I have found a set of open source gerber files which make this issue reproducible. The files can be downloaded from https://oshpark.com/shared_projects/ztyE14yF -- just in case they will be removed I've added them to the issue too. 

Any thoughts on how to solve this?

This is how OSH park draws the image:

[image]

And this is how pcb-tools does it:

[image]

Files: [bbbc6260f32452045be1159cf18f7d95.zip](https://github.com/curtacircuitos/pcb-tools/files/1450617/bbbc6260f32452045be1159cf18f7d95.zip)

In this case the silkscreen is also shown outside the outline. I don't mind if that's not visible on the exported image of course.

chintal, 2019-05-11:
@ju5t Silk screen overhangs don't typically count within board bounds. Gerber files contain an outline layer which does. Usually, these overhangs relate to connectors and such which aren't supposed to be a part of board size, and need special consideration on part of the mechanical housing design.

Usually, when looking at board bounds, you typically want to see the envelope within which the unpopulated board itself fits. 

It may well be possible to use layer_bounds for each layer and develop a board_extent which does include the silk screen overhang, but in practice this can be a complicated measurement to get right.

ju5t, 2019-05-11:
> Usually, these overhangs relate to connectors and such which aren't supposed to be a part of board size, and need special consideration on part of the mechanical housing design.

As I said, I am not really after the missing silkscreen. Although I do believe that if you're rendering an image that is meant to reflect the design, it should be included. You can't leave it out.

The problem is the outline itself.

If you open the zip file you will see `BLDC_4-Edge_Cuts.gm1`. This is what it should use to calculate the board size but yet in the image it is cut off. Not only at the bottom, also at the sides. Even copper that is within bounds of the outline is stripped away.

chintal, 2019-05-13:
I see. In case someone else searches and finds this :

Yes, it uses the top copper layer to determine the bounds if there isn't an outline layer available. In general, there should be an outline layer available. Depending on the software used, there are often instances where the gerber set is incomplete, when the PCB is designed enough for the manufacturer to fill in the details. For some other softwares, it is possible the outline layer is not correctly identified by pcb-tools, in which case an issue should be filed.

ju5t, 2019-05-26:
> in which case an issue should be filed.

Huh? Maybe I am missing your point but that is exactly why I raised this issue. What do you mean?

chintal, 2019-05-27:
I was speaking generally since you said you weren't using this anymore. 

That said, What I mean is that a relatively easily fixable issue would be if the outline layer is not correctly detected but should be for whatever EDA software you're using, such as if <>-Edge_Cuts.gm1 is where it usually places the outline. The code currently has a hint that recognizes "Edge.Cuts", not "Edge_Cuts" (https://github.com/curtacircuitos/pcb-tools/blob/2601ae8eab8d7be807bdbed264cd943e441a8da0/gerber/layers.py#L87) .