curtacircuitos/pcb-tools #72: Example fails

greenscreenflicker, 2017-06-11:
I followed the installation steps. However the example fails.

michael@oldgrey:~/work/pcb-tools/examples$ python render_gerbv_tests.py 
Loaded Gerber file: test-aperture-circle-1.gbx
Traceback (most recent call last):
  File "render_gerbv_tests.py", line 52, in <module>
    layer.render(ctx)
  File "build/bdist.linux-x86_64/egg/gerber/cam.py", line 274, in render
    ctx._paint_background()
  File "build/bdist.linux-x86_64/egg/gerber/render/cairo_backend.py", line 579, in _paint_background
    self.output_ctx.set_source_rgba(*color, alpha=alpha)
TypeError: set_source_rgba() takes no keyword arguments
Loaded Gerber file: test-image-rotation-1.gbx
Traceback (most recent call last):
  File "render_gerbv_tests.py", line 52, in <module>
    layer.render(ctx)
  File "build/bdist.linux-x86_64/egg/gerber/cam.py", line 274, in render
    ctx._paint_background()
  File "build/bdist.linux-x86_64/egg/gerber/render/cairo_backend.py", line 579, in _paint_background
    self.output_ctx.set_source_rgba(*color, alpha=alpha)
TypeError: set_source_rgba() takes no keyword arguments
Loaded Gerber file: test-layer-knockout-1.gbx
Traceback (most recent call last):
  File "render_gerbv_tests.py", line 52, in <module>
    layer.render(ctx)
  File "build/bdist.linux-x86_64/egg/gerber/cam.py", line 274, in render
    ctx._paint_background()
  File "build/bdist.linux-x86_64/egg/gerber/render/cairo_backend.py", line 579, in _paint_background
    self.output_ctx.set_source_rgba(*color, alpha=alpha)
TypeError: set_source_rgba() takes no keyword arguments
Loaded Gerber file: test-image-polarity-1.gbx
Traceback (most recent call last):
  File "render_gerbv_tests.py", line 52, in <module>
    layer.render(ctx)
  File "build/bdist.linux-x86_64/egg/gerber/cam.py", line 274, in render
    ctx._paint_background()
  File "build/bdist.linux-x86_64/egg/gerber/render/cairo_backend.py", line 579, in _paint_background
    self.output_ctx.set_source_rgba(*color, alpha=alpha)
TypeError: set_source_rgba() takes no keyword arguments
Loaded Gerber file: test-aperture-polygon-flash-1.gbx
Traceback (most recent call last):
  File "render_gerbv_tests.py", line 52, in <module>
    layer.render(ctx)
  File "build/bdist.linux-x86_64/egg/gerber/cam.py", line 274, in render
    ctx._paint_background()
  File "build/bdist.linux-x86_64/egg/gerber/render/cairo_backend.py", line 579, in _paint_background
    self.output_ctx.set_source_rgba(*color, alpha=alpha)
TypeError: set_source_rgba() takes no keyword arguments
Loaded Gerber file: test-layer-step-and_repeat-1.gbx
Traceback (most recent call last):
  File "render_gerbv_tests.py", line 52, in <module>
    layer.render(ctx)
  File "build/bdist.linux-x86_64/egg/gerber/cam.py", line 274, in render
    ctx._paint_background()
  File "build/bdist.linux-x86_64/egg/gerber/render/cairo_backend.py", line 579, in _paint_background
    self.output_ctx.set_source_rgba(*color, alpha=alpha)
TypeError: set_source_rgba() takes no keyword arguments
Loaded Gerber file: test-aperture-polygon-1.gbx
Traceback (most recent call last):
  File "render_gerbv_tests.py", line 52, in <module>
    layer.render(ctx)
  File "build/bdist.linux-x86_64/egg/gerber/cam.py", line 274, in render
    ctx._paint_background()
  File "build/bdist.linux-x86_64/egg/gerber/render/cairo_backend.py", line 579, in _paint_background
    self.output_ctx.set_source_rgba(*color, alpha=alpha)
TypeError: set_source_rgba() takes no keyword arguments
Loaded Gerber file: test-layer-scale-factor-1.gbx
Traceback (most recent call last):
  File "render_gerbv_tests.py", line 52, in <module>
    layer.render(ctx)
[... 180 lines cut ...]
  File "build/bdist.linux-x86_64/egg/gerber/cam.py", line 274, in render
    ctx._paint_background()
  File "build/bdist.linux-x86_64/egg/gerber/render/cairo_backend.py", line 579, in _paint_background
    self.output_ctx.set_source_rgba(*color, alpha=alpha)
TypeError: set_source_rgba() takes no keyword arguments
Loaded Gerber file: test-image-offset-2.gbx
Traceback (most recent call last):
  File "render_gerbv_tests.py", line 52, in <module>
    layer.render(ctx)
  File "build/bdist.linux-x86_64/egg/gerber/cam.py", line 274, in render
    ctx._paint_background()
  File "build/bdist.linux-x86_64/egg/gerber/render/cairo_backend.py", line 579, in _paint_background
    self.output_ctx.set_source_rgba(*color, alpha=alpha)
TypeError: set_source_rgba() takes no keyword arguments
Loaded Gerber file: test-layer-offset-1.gbx
Traceback (most recent call last):
  File "render_gerbv_tests.py", line 52, in <module>
    layer.render(ctx)
  File "build/bdist.linux-x86_64/egg/gerber/cam.py", line 274, in render
    ctx._paint_background()
  File "build/bdist.linux-x86_64/egg/gerber/render/cairo_backend.py", line 579, in _paint_background
    self.output_ctx.set_source_rgba(*color, alpha=alpha)
TypeError: set_source_rgba() takes no keyword arguments

How to fix?

greenscreenflicker, 2017-06-12:
@curtacircuitos

hamiltonkibbe, 2017-06-12:
This should be fixed by f7a719e6f7 can you try again with the latest version

greenscreenflicker, 2017-06-13:
@hamiltonkibbe 
I'm afraid not all fixed.

michael@oldgrey:~/work/pcb-tools/examples$ python cairo_example.py 
Traceback (most recent call last):
  File "cairo_example.py", line 44, in <module>
    ctx.render_layer(copper)
  File "build/bdist.linux-x86_64/egg/gerber/render/cairo_backend.py", line 98, in render_layer
  File "build/bdist.linux-x86_64/egg/gerber/render/cairo_backend.py", line 198, in _render_layer
  File "build/bdist.linux-x86_64/egg/gerber/render/render.py", line 154, in render
  File "build/bdist.linux-x86_64/egg/gerber/render/cairo_backend.py", line 363, in _render_rectangle
TypeError: rectangle() takes no keyword arguments

michael@oldgrey:~/work/pcb-tools/examples$ python render_gerbv_tests.py 
Loaded Gerber file: test-aperture-circle-1.gbx
Traceback (most recent call last):
  File "render_gerbv_tests.py", line 52, in <module>
    layer.render(ctx)
  File "build/bdist.linux-x86_64/egg/gerber/cam.py", line 278, in render
    ctx.render(p)
  File "build/bdist.linux-x86_64/egg/gerber/render/render.py", line 148, in render
    self._render_arc(primitive, color)
  File "build/bdist.linux-x86_64/egg/gerber/render/cairo_backend.py", line 255, in _render_arc
    mask.ctx.arc(*center, radius=radius, angle1=angle1, angle2=angle2)
TypeError: arc() takes no keyword arguments
Loaded Gerber file: test-image-rotation-1.gbx
Writing output to: test-image-rotation-1.png
Loaded Gerber file: test-layer-knockout-1.gbx
Writing output to: test-layer-knockout-1.png
Loaded Gerber file: test-image-polarity-1.gbx
Writing output to: test-image-polarity-1.png
Loaded Gerber file: test-aperture-polygon-flash-1.gbx
Writing output to: test-aperture-polygon-flash-1.png
Loaded Gerber file: test-layer-step-and_repeat-1.gbx
Writing output to: test-layer-step-and_repeat-1.png
Loaded Gerber file: test-aperture-polygon-1.gbx
Traceback (most recent call last):
  File "render_gerbv_tests.py", line 52, in <module>
    layer.render(ctx)
  File "build/bdist.linux-x86_64/egg/gerber/cam.py", line 278, in render
    ctx.render(p)
  File "build/bdist.linux-x86_64/egg/gerber/render/render.py", line 148, in render
    self._render_arc(primitive, color)
  File "build/bdist.linux-x86_64/egg/gerber/render/cairo_backend.py", line 255, in _render_arc
    mask.ctx.arc(*center, radius=radius, angle1=angle1, angle2=angle2)
TypeError: arc() takes no keyword arguments
Loaded Gerber file: test-layer-scale-factor-1.gbx
Writing output to: test-layer-scale-factor-1.png
Loaded Gerber file: test-image-offset-1.gbx
Writing output to: test-image-offset-1.png
Loaded Gerber file: test-aperture-rectangle-1.gbx
Traceback (most recent call last):
  File "render_gerbv_tests.py", line 52, in <module>
    layer.render(ctx)
  File "build/bdist.linux-x86_64/egg/gerber/cam.py", line 278, in render
    ctx.render(p)
  File "build/bdist.linux-x86_64/egg/gerber/render/render.py", line 148, in render
    self._render_arc(primitive, color)
  File "build/bdist.linux-x86_64/egg/gerber/render/cairo_backend.py", line 255, in _render_arc
    mask.ctx.arc(*center, radius=radius, angle1=angle1, angle2=angle2)
TypeError: arc() takes no keyword arguments
Loaded Gerber file: test-layer-rotation-1.gbx
Writing output to: test-layer-rotation-1.png
Loaded Gerber file: test-polygon-fill-1.gbx
Traceback (most recent call last):
  File "render_gerbv_tests.py", line 52, in <module>
    layer.render(ctx)
  File "build/bdist.linux-x86_64/egg/gerber/cam.py", line 278, in render
    ctx.render(p)
  File "build/bdist.linux-x86_64/egg/gerber/render/render.py", line 150, in render
[... 93 lines cut ...]
    self._render_circle(circle, color)
  File "build/bdist.linux-x86_64/egg/gerber/render/cairo_backend.py", line 315, in _render_circle
    angle2=(2 * math.pi))
TypeError: arc() takes no keyword arguments
Loaded Gerber file: test-layer-knockout-2.gbx
Writing output to: test-layer-knockout-2.png
Loaded Excellon file: test-drill-repeat-1.exc
Traceback (most recent call last):
  File "render_gerbv_tests.py", line 52, in <module>
    layer.render(ctx)
  File "build/bdist.linux-x86_64/egg/gerber/cam.py", line 278, in render
    ctx.render(p)
  File "build/bdist.linux-x86_64/egg/gerber/render/render.py", line 160, in render
    self._render_drill(primitive, self.color)
  File "build/bdist.linux-x86_64/egg/gerber/render/cairo_backend.py", line 510, in _render_drill
    self._render_circle(circle, color)
  File "build/bdist.linux-x86_64/egg/gerber/render/cairo_backend.py", line 315, in _render_circle
    angle2=(2 * math.pi))
TypeError: arc() takes no keyword arguments
Loaded Gerber file: test-layer-axis-select-1.gbx
Writing output to: test-layer-axis-select-1.png
Loaded Gerber file: test-image-offset-2.gbx
Writing output to: test-image-offset-2.png
Loaded Gerber file: test-layer-offset-1.gbx
Writing output to: test-layer-offset-1.png

hamiltonkibbe, 2017-07-04:
removed the rest of the instances of mixing unpacks with kwargs... Let me know if that fixes it for you.

commit e754f594 (2017-07-04): Remove rest of mixed unpack/kwarg syntax to fix #72