curtacircuitos/pcb-tools #74: Running tests in travis fails

kliment, 2017-09-18:
I forked the repo and tried running the travis CI tests on the current master, and the tests fail although it's the exact same commit. [This](https://travis-ci.org/curtacircuitos/pcb-tools/builds/249874328) build from 3 months ago passes but [this](https://travis-ci.org/kliment/pcb-tools/builds/276752212) build of the same commit from today fails. Any idea what's happening?

kliment, 2017-09-18:
Looks like it's related to https://blog.travis-ci.com/2017-08-31-trusty-as-default-status because setting the dist to precise makes it work

hamiltonkibbe, 2017-09-22:
It appears to be due to differing versions of Cairo. It's likely that the naive image file comparison we're doing in those tests that are failing will have to be replaced with something more robust