curtacircuitos/pcb-tools #77: TypeError: 'instancemethod' object is not iterable

ju5t, 2017-11-01:
I haven't had the chance to dig into this yet but either way a bug report is in place. I'm getting the following error:

```
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/celery/app/trace.py", line 374, in trace_task
    R = retval = fun(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/celery/app/trace.py", line 629, in __protected_call__
    return self.run(*args, **kwargs)
  File "/home/app/pcb-tools/tasks.py", line 94, in convert_gerbers
    max_width=max_width, max_height=max_height)
  File "/home/app/pcb-tools/gerber/render/cairo_backend.py", line 130, in render_layers
    verbose=verbose)
  File "/home/app/pcb-tools/gerber/render/cairo_backend.py", line 98, in render_layer
    self._render_layer(layer, settings)
  File "/home/app/pcb-tools/gerber/render/cairo_backend.py", line 198, in _render_layer
    self.render(prim)
  File "/home/app/pcb-tools/gerber/render/render.py", line 162, in render
    self._render_slot(primitive, self.color)
  File "/home/app/pcb-tools/gerber/render/cairo_backend.py", line 521, in _render_slot
    with self._clip_primitive(slot):
  File "/home/app/pcb-tools/gerber/render/cairo_backend.py", line 630, in _clip_primitive
    return Clip(primitive)
  File "/home/app/pcb-tools/gerber/render/cairo_backend.py", line 602, in __init__
    x_range, y_range = primitive.bounding_box
TypeError: 'instancemethod' object is not iterable
```

Unfortunately I can't share the gerber files. I don't see the same issues with other files.

hamiltonkibbe, 2017-11-01:
@ju5t Should be fixed by e12a04f, confirm and i'll close

ju5t, 2017-11-02:
@hamiltonkibbe it does solve this specific error. It gets one step further now and says:

```
  File "/home/api/app/conversions/gerber/render/cairo_backend.py", line 602, in __init__
    x_range, y_range = primitive.bounding_box
  File "/home/app/pcb-tools/gerber/primitives.py", line 1672, in bounding_box
    ll = tuple([c - self.diameter / 2. for c in self.position])
AttributeError: 'Slot' object has no attribute 'position'
```

ju5t, 2017-11-09:
@hamiltonkibbe I've tried looking into this myself but it surpasses my knowledge of gerber files.

It looks like Slot is defined on the following line: https://github.com/curtacircuitos/pcb-tools/blob/7ad6c3f6acfe1fe995c9f087e7ef7a51add60afe/gerber/excellon.py#L218

And on line:
https://github.com/curtacircuitos/pcb-tools/blob/e12a04fc16fd3c43a1353658a528ac8325ef42bb/gerber/primitives.py#L1672

It tries to get the position attribute which is a property on other classes -- and that returns `_position`, but that's never set.

https://github.com/curtacircuitos/pcb-tools/blob/e12a04fc16fd3c43a1353658a528ac8325ef42bb/gerber/primitives.py#L1655-L1662

Is there anything I can do to help here?

hamiltonkibbe, 2017-11-09:
I've been pretty busy this week, i'll probably be able to look at this on Friday or saturday, I have a pretty good idea of what the issue is.

hamiltonkibbe, 2017-11-14:
@ju5t give the fix-slot-position branch a try

ju5t, 2017-11-16:
@hamiltonkibbe this solves the issue for me. Awesome, thanks!

hamiltonkibbe, 2017-11-16:
merged. I'll take a look at that other PR, haven't had a chance to look at it yet

closed 2017-11-16 by hamiltonkibbe