curtacircuitos/pcb-tools #95: question: gerber program?

crazyjc23, 2019-02-25:
what program and version should be used to for output files and what file extensions does it support. i have tried kicad and eagle but the files the output files are not recognized by pcbtools

hamiltonkibbe, 2019-02-28:
It should work with gerber/excellon files exported from either program, from eagle you need to use a CAM job to  export the gerber/excellon files. The file extensions from the default 2/4 layer CAM Job outputs should be fine.

crazyjc23, 2019-02-28:
I think i found the problem, i tried and was successful with the python on windows but when using raspberry pi, it always says dump error. Could you help please help me fixing that

crazyjc23, 2019-02-28:
I tracked down the issue and it is in cairo_backend.py line 250, retun self.surface.write_to_png(filename)

Somehow raspberry pi always throws in a memory error in this part.

hamiltonkibbe, 2019-03-02:
Are the gerber files you're trying to render particularly large?
If you provide a stack trace (if available) that may give some more direction as to where to look.

crazyjc23, 2019-03-02:
It's a 5by5 inch circuit design extracted from eagle. And im sorry im pretty much a newb in this.

crazyjc23, 2019-03-02:
Im trying to use it for my school project. It uses a raspberry pi. The circuit design is a single sided 5by5 inch circuit. I tried reinstalling the cairocffi to the latest version but cant still cant get it to work

hamiltonkibbe, 2019-03-02:
if you can send me the gerber files I can try to reproduce on a pi later this weekend.
What os and python version are you using?