curtacircuitos/pcb-tools #96: Rendering/Font Issue on Aws

adithyadinesh96, 2019-03-02:
Hi,

We were using your old source(may be 2yrs old) for some time, to render. We recently updated to your new source (rendering has improved a lot, in Ubuntu)
But the issue is running on AWS EC2 Instance (Amazon Linux 1 OS). Old source used to run but the new source is not able to render properly. There are no errors but wont render properly and not sure what libraries are causing this issue.
We are using master branch. We tried rendering with sample gerber files present in repo.
Attached are the rendered images on aws.
 (https://github.com/curtacircuitos/pcb-tools/tree/master/examples/gerbers)
[pcb_images.zip](https://github.com/curtacircuitos/pcb-tools/files/2921872/pcb_images.zip)

skrajath, 2019-04-01:
Hi,

I am too facing issues on AWS EB, is there a solution to this??
What am I missing here? Libraries/dependencies ate not getting installed properly .

hamiltonkibbe, 2019-04-01:
Can you share the output of `pip freeze` and the version of cairo you have installed system-wide?

skrajath, 2019-04-01:
So in the Virtual env
`pip freeze` has these

```
cairocffi==0.6
cffi==1.12.2
pycparser==2.19
```

And the version of cairo you have installed system-wide is this

`1.12.14-6.8.amzn1`

Let me know if you need any more details, I will get it right away.

xiaoqiangnobug, 2020-07-22:
Hello,

I have the same problem. Have you solved your problem?