curtacircuitos/pcb-tools #98: GerberCairoContext import error

noyart, 2019-04-10:
Hi! 
Im trying to set everyhing up and try the exampel code. Im new to python. 
I get this error when I run the exampel code. What can be the reason for the import error of the GerberCairoContext. Wrong version somewhere? I run a windows machine with python 3.

Exception has occurred: ImportError
cannot import name 'GerberCairoContext' from 'gerber.render' (C:\Users\Noy\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pcb_tools-0.1.6-py3.7.egg\gerber\render\_init_.py)
  File "G:\phytonscrips\AOI\gerbertest.py", line 2, in <module>
    from gerber.render import GerberCairoContext
  File "C:\Users\Noy\AppData\Local\Programs\Python\Python37-32\Lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\Noy\AppData\Local\Programs\Python\Python37-32\Lib\runpy.py", line 96, in _run_module_code
    mod_name, mod_spec, pkg_name, script_name)
  File "C:\Users\Noy\AppData\Local\Programs\Python\Python37-32\Lib\runpy.py", line 263, in run_path
    pkg_name=pkg_name, script_name=fname)

hamiltonkibbe, 2019-04-10:
Think it needs to be imported from cairo_backend:

`from gerber.render.cairo_backend import GerberCairoContext`

noyart, 2019-04-14:
Sorry for the late reply, I still get the same problem that it cant import on that line. 
I must have done something wrong during the installation right?

phsilva, 2019-11-26:
Fixed on master, Thanks for the report.

closed 2019-11-26 by phsilva

kahlju, 2019-12-16:
This is not fixed.  please open it back up.  @chintal ruined the master in his latest commit to overwrite all the changes in the commit where @phsilva fixed all of this.  I'm not sure how he/she ran GerberCairoContext with his code, but it does not work again.  I'm reinstalling it with the "correct" code version which is not the master file anymore.

alvinsu51416, 2020-01-11:
Agreed with @kahlju, the new update does not fix the issue. I got the same error like @noyart did with pcb-tools 0.1.6

senic35, 2020-11-06:
Is there a fix to this? The issue still persists and what hamiltonkibbe suggested doesn't work for me, I get the following error: `OSError: dlopen() failed to load a library: cairo / cairo-2 / cairo-gobject-2`

vladiks, 2020-12-03:
I have the same problem as @noyart  `No module named 'gerber.render'; 'gerber' is not a package`

First line of example `import gerber` works, but running `print(dir(gerber))` shows only `['__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__spec__', 'gerber']` , so there is for example no read method

vladiks, 2020-12-04:
> I have the same problem as @noyart `No module named 'gerber.render'; 'gerber' is not a package`
>
> First line of example `import gerber` works, but running `print(dir(gerber))` shows only [...]

Ok, so I found out, that i named my python file gerber.py, so `import gerber` was actually importing my own file :-)

I renamed the file and used the import suggested by @hamiltonkibbe and it got me to "cannot load library 'libcairo.so.2'", so that is different issue

wytr, 2021-11-10:
Bump, i just got the same Error which does not seem to be fixed by any advice given here.

pablo-johnson, 2022-03-03:
For the ones using docker, I put this command in the Dockerfile and it solved the problem 
`RUN apt-get install -y libcairo2`

If you are not using docker, I supposed that it is solved just by running `apt-get install -y libcairo2` in the terminal.

BrewingGeek, 2023-07-12:
'apt-get install -y libcairo2' my work for a Linux box but how about a Windows box...

ImportError: cannot import name 'GerberCairoContext' from 'gerber.render'

commit 404384cf (2019-11-26): Fix #98
commit 6b9cc248 (2019-11-26): Merge branch 'master' into ph-move-to-pytest

* master:
  Fix #98
  Run workflow checks on PRs
  Bump sphinx from 1.2.3 to 2.2.1
  Bump numpydoc from 0.5 to 0.9.1
  Add hook for outline layer to PCB class
  Add a new transparant theme for multilayer renders.