curtacircuitos/pcb-tools #133: CairoBackend.render_layer incorrectly uses layer.bounds instead of layer.bounding_box

jaseg, 2020-03-02:
When rendering a layer to a Cairo surface, CairoBackend uses the layer's `bounds` attribute to determine surface size. This is incorrect and `bounding_box` should be used instead.

`bounds` will only take into account the endpoints of gerber primitives. This is fine for straight lines but fails in some cases for e.g. arcs. Take a board shaped like a square with an arc protruding on one side like this:
```
+-----+
|     |
|     )
|     |
+-----+
```
In this case if the protrusion on the right is parametrized using only a single arc, the computed bounds will be too small to contain the entire board.

A possible test case would be a full circle (whose bounds I guess will yield (0, 0)).

I'd contribute a PR myself but I don't have the time rn to work myself into the code base enough to patch this without breaking something else.

This may be the cause for #90

phsilva, 2020-03-02:
Thanks @jaseg, will take a look into this and #90 asap.

cofactr-mh, 2021-07-26:
Any update on this?