curtacircuitos/pcb-tools #142: run cairo_example.py got fuzzy rendered images

dafeiyoung, 2020-04-17:
Hello， everyone!
I ran _cairo_example.py_ successfully but I got fuzzy rendered images.

Here's what I have done:
After downloaded pcb-tools source code and installed all requirements, I just tryied to run _pcb-tools-master/examples/cairo_example.py_ and no error occurred. It was supposed to get images like these:
[image]
[image]

However, I got images like these:
[image]
[image]

It seems that these images was incorrectly rendered. As you can see, these images are fuzzy and some details were lost.

Is someone know what's going on?

My environment: ubuntu14.04x64 & python2.7.6(tried python3.6.4 but it didn't work either)
My friend who's using ubuntu16.04 and python3.6.4 DOES NOT facing the same problem. Is it related to my ubuntu14.04?

dafeiyoung, 2020-04-17:
BTW: the source code was downloaded from master branch.

LEVEN08069, 2020-05-09:
老哥可以加你的扣扣,问问你这个问题解决了吗？