curtacircuitos/pcb-tools #192: Compatibility issue with windows

callowidealist, 2021-01-11:
I tried running it on my windows machine but got multiple errors related to packages , though i figured out them but i now i'm having multiple syntax error !!

`PS E:\pcb-tools>  e:; cd 'e:\pcb-tools'; & 'C:\Users\raghavendra_tripathi\AppData\Local\Programs\Python\Python39\python.exe' 'c:\Users\raghavendra_tripathi\.vscode\extensions\ms-python.python-2020.12.424452561\pythonFiles\lib\python\debugpy\launcher' '60020' '--' 'e:\pcb-tools\examples\cairo_example.py'
C:\Users\raghavendra_tripathi\AppData\Local\Programs\Python\Python39\lib\site-packages\pcb_tools-0.1-py3.9.egg\gerber\excellon_statements.py:458: 
SyntaxWarning: "is not" with a literal. Did you mean "!="?
C:\Users\raghavendra_tripathi\AppData\Local\Programs\Python\Python39\lib\site-packages\pcb_tools-0.1-py3.9.egg\gerber\excellon_statements.py:461: 
SyntaxWarning: "is not" with a literal. Did you mean "!="?
C:\Users\raghavendra_tripathi\AppData\Local\Programs\Python\Python39\lib\site-packages\pcb_tools-0.1-py3.9.egg\gerber\excellon_statements.py:609: 
SyntaxWarning: "is not" with a literal. Did you mean "!="?
C:\Users\raghavendra_tripathi\AppData\Local\Programs\Python\Python39\lib\site-packages\pcb_tools-0.1-py3.9.egg\gerber\excellon_statements.py:612: 
SyntaxWarning: "is not" with a literal. Did you mean "!="?
C:\Users\raghavendra_tripathi\AppData\Local\Programs\Python\Python39\lib\site-packages\pcb_tools-0.1-py3.9.egg\gerber\excellon_statements.py:458: 
SyntaxWarning: "is not" with a literal. Did you mean "!="?
C:\Users\raghavendra_tripathi\AppData\Local\Programs\Python\Python39\lib\site-packages\pcb_tools-0.1-py3.9.egg\gerber\excellon_statements.py:461: 
SyntaxWarning: "is not" with a literal. Did you mean "!="?
C:\Users\raghavendra_tripathi\AppData\Local\Programs\Python\Python39\lib\site-packages\pcb_tools-0.1-py3.9.egg\gerber\excellon_statements.py:609: 
SyntaxWarning: "is not" with a literal. Did you mean "!="?
C:\Users\raghavendra_tripathi\AppData\Local\Programs\Python\Python39\lib\site-packages\pcb_tools-0.1-py3.9.egg\gerber\excellon_statements.py:612: 
SyntaxWarning: "is not" with a literal. Did you mean "!="?
C:\Users\raghavendra_tripathi\AppData\Local\Programs\Python\Python39\lib\site-packages\pcb_tools-0.1-py3.9.egg\gerber\ipc356.py:385: SyntaxWarning: "is not" with a literal. Did you mean "!="?
C:\Users\raghavendra_tripathi\AppData\Local\Programs\Python\Python39\lib\site-packages\pcb_tools-0.1-py3.9.egg\gerber\ipc356.py:386: SyntaxWarning: "is not" with a literal. Did you mean "!="?
C:\Users\raghavendra_tripathi\AppData\Local\Programs\Python\Python39\lib\site-packages\pcb_tools-0.1-py3.9.egg\gerber\ipc356.py:415: SyntaxWarning: "is not" with a literal. Did you mean "!="?
C:\Users\raghavendra_tripathi\AppData\Local\Programs\Python\Python39\lib\site-packages\pcb_tools-0.1-py3.9.egg\gerber\ipc356.py:417: SyntaxWarning: "is not" with a literal. Did you mean "!="?
C:\Users\raghavendra_tripathi\AppData\Local\Programs\Python\Python39\lib\site-packages\pcb_tools-0.1-py3.9.egg\gerber\ipc356.py:431: SyntaxWarning: "is not" with a literal. Did you mean "!="?
C:\Users\raghavendra_tripathi\AppData\Local\Programs\Python\Python39\lib\site-packages\pcb_tools-0.1-py3.9.egg\gerber\ipc356.py:432: SyntaxWarning: "is not" with a literal. Did you mean "!="?
C:\Users\raghavendra_tripathi\AppData\Local\Programs\Python\Python39\lib\site-packages\pcb_tools-0.1-py3.9.egg\gerber\ipc356.py:385: SyntaxWarning: "is not" with a literal. Did you mean "!="?
C:\Users\raghavendra_tripathi\AppData\Local\Programs\Python\Python39\lib\site-packages\pcb_tools-0.1-py3.9.egg\gerber\ipc356.py:386: SyntaxWarning: "is not" with a literal. Did you mean "!="?
C:\Users\raghavendra_tripathi\AppData\Local\Programs\Python\Python39\lib\site-packages\pcb_tools-0.1-py3.9.egg\gerber\ipc356.py:415: SyntaxWarning: "is not" with a literal. Did you mean "!="?
C:\Users\raghavendra_tripathi\AppData\Local\Programs\Python\Python39\lib\site-packages\pcb_tools-0.1-py3.9.egg\gerber\ipc356.py:417: SyntaxWarning: "is not" with a literal. Did you mean "!="?
C:\Users\raghavendra_tripathi\AppData\Local\Programs\Python\Python39\lib\site-packages\pcb_tools-0.1-py3.9.egg\gerber\ipc356.py:431: SyntaxWarning: "is not" with a literal. Did you mean "!="?
C:\Users\raghavendra_tripathi\AppData\Local\Programs\Python\Python39\lib\site-packages\pcb_tools-0.1-py3.9.egg\gerber\ipc356.py:432: SyntaxWarning: "is not" with a literal. Did you mean "!="?

`

gabrielkarras, 2021-01-15:
Hi!

I had the same issue. I believe it's because we're using python 3, while the developers are using python 2? But I'm still having the same issues with python 2. Although, that could simply be my own fault.

Hopefully, the contributors would know how to tackle this issue.

sybrenstuvel, 2021-02-14:
The code has a bunch of these constructs:

```
x = int(coord_dict['x']) if coord_dict['x'] is not '' else x
```

The `is not` here is incorrect, and also was back in those ancient Python 2.7 times. The only reason it works is that CPython has certain optimizations when it comes to short strings. Since empty strings are falsy, all `is not ''` can just be removed, resulting in code like this:

```
x = int(coord_dict['x']) if coord_dict['x'] else x
```

byteptr, 2021-12-26:
I'm wondering how these issue is closed while warnings still remains. 
I replaced all 'is not ' with '!=' in the excellont_statements.py and ipc365.py and the warnings no longer exists.

[changes.zip](https://github.com/curtacircuitos/pcb-tools/files/7777377/changes.zip)