curtacircuitos/pcb-tools #211: Parsing gerber fails with "ValueError: could not convert string to float: 'OMP137'"

positron96, 2021-05-25:
The code:
```
import gerber
top_copper = gerber.read(r'D:\Reports\combined.gtl')
```

Produces error:
```
ValueError                                Traceback (most recent call last)
<ipython-input-4-62a30abcaa4c> in <module>
----> 1 top_copper = gerber.read(r'D:\Reports\combined.gtl')

d:\DATA\Documents\Programming\python\lib\site-packages\gerber\common.py in read(filename)
     39     with open(filename, 'rU') as f:
     40         data = f.read()
---> 41     return loads(data, filename)
     42 
     43 

d:\DATA\Documents\Programming\python\lib\site-packages\gerber\common.py in loads(data, filename)
     63     fmt = detect_file_format(data)
     64     if fmt == 'rs274x':
---> 65         return rs274x.loads(data, filename=filename)
     66     elif fmt == 'excellon':
     67         return excellon.loads(data, filename=filename)

d:\DATA\Documents\Programming\python\lib\site-packages\gerber\rs274x.py in loads(data, filename)
     68         A GerberFile created from the specified file.
     69     """
---> 70     return GerberParser().parse_raw(data, filename)
     71 
     72 

d:\DATA\Documents\Programming\python\lib\site-packages\gerber\rs274x.py in parse_raw(self, data, filename)
    267     def parse_raw(self, data, filename=None):
    268         self.filename = filename
--> 269         for stmt in self._parse(self._split_commands(data)):
    270             self.evaluate(stmt)
    271             self.statements.append(stmt)

d:\DATA\Documents\Programming\python\lib\site-packages\gerber\rs274x.py in _parse(self, data)
    376                         yield LPParamStmt.from_dict(param)
    377                     elif param["param"] == "AD":
--> 378                         yield ADParamStmt.from_dict(param)
    379                     elif param["param"] == "AM":
    380                         stmt = AMParamStmt.from_dict(param)

d:\DATA\Documents\Programming\python\lib\site-packages\gerber\gerber_statements.py in from_dict(cls, stmt_dict)
    320         shape = stmt_dict.get('shape')
    321         modifiers = stmt_dict.get('modifiers')
--> 322         return cls(param, d, shape, modifiers)
    323 
    324     def __init__(self, param, d, shape, modifiers):

d:\DATA\Documents\Programming\python\lib\site-packages\gerber\gerber_statements.py in __init__(self, param, d, shape, modifiers)
    351             self.modifiers = modifiers
    352         elif modifiers:
--> 353             self.modifiers = [tuple([float(x) for x in m.split("X") if len(x)])
    354                               for m in modifiers.split(",") if len(m)]
    355         else:

d:\DATA\Documents\Programming\python\lib\site-packages\gerber\gerber_statements.py in <listcomp>(.0)
    351             self.modifiers = modifiers
    352         elif modifiers:
--> 353             self.modifiers = [tuple([float(x) for x in m.split("X") if len(x)])
    354                               for m in modifiers.split(",") if len(m)]
    355         else:

d:\DATA\Documents\Programming\python\lib\site-packages\gerber\gerber_statements.py in <listcomp>(.0)
    351             self.modifiers = modifiers
    352         elif modifiers:
--> 353             self.modifiers = [tuple([float(x) for x in m.split("X") if len(x)])
    354                               for m in modifiers.split(",") if len(m)]
    355         else:

ValueError: could not convert string to float: 'OMP137'

```

Here is the file in question: [combined.gtl.txt](https://github.com/curtacircuitos/pcb-tools/files/6536787/combined.gtl.txt)

The error might be at line 71, but it's hard to tell. The file is opened successfully with online gerber viewers.

bhclowers, 2022-03-04:
I'm running into a similar issue.  Did you ever identify a solution?

positron96, 2022-03-04:
Nope. 
I got the impression that the library simply doesn't support this feature. Someone with deep knowledge of Gerber needs to implement it.