curtacircuitos/pcb-tools #222: pcb-tools draw a wrong way?

jekoie, 2022-06-09:
Below gerber file, why pcb-tools draw it in a wrong way?
```console
G04 This file illustrates how to use levels to create holes*
%FSLAX25Y25*%
%MOMM*%
G01*
G04 First level: big square - dark polarity*
%LPD*%
G36*
X250000Y250000D02*
X1750000D01*
Y1750000D01*
X250000D01*
Y250000D01*
G37*
G04 Second level: big circle - clear polarity*
%LPC*%
G36*
G75*
X500000Y1000000D02*
G03*
X500000Y1000000I500000J0D01*
G37*
G04 Third level: small square - dark polarity*
%LPD*%
G36*
X750000Y750000D02*
X1250000D01*
Y1250000D01*
X750000D01*
Y750000D01*
G37*
G04 Fourth level: small circle - clear polarity*
%LPC*%
G36*
G75*
X1150000Y1000000D02*
G03*
X1150000Y1000000I250000J0D01*
G37*
M02*
```
It should be this:
[image]

pcb-tools draw this:
[image]

Argmaster, 2023-09-11:
Hi, code you have provided doesn't create apertures nor select apertures which is a bit unexpected. However, this file is a valid Gerber file and should create what you have shown as expected.

Argmaster, 2023-09-11:
Is it possible that you are missing G01 code somewhere around `G04 Third level: small square - dark polarity*`? As you are not setting line drawing mode before attempting to draw that square.

jekoie, 2023-09-12:
I don't know G01, the code I provided is from some where from internet.