curtacircuitos/pcb-tools #227: poking curtacircuitos

rusefillc, 2023-03-17:
Hello from 2023 :)

Just checking if @curtacircuitos is around?

We use this pcb-tools for rusEFI open source engine control unit https://github.com/andreika-git/hellen-one/ and we LOVE it!

Looks like there are some cool pull requests without attention?

For instance https://github.com/curtacircuitos/pcb-tools/pull/225 and https://github.com/curtacircuitos/pcb-tools/pull/208 is there a hope to get progress on those?

phsilva, 2023-04-06:
Hey @rusefillc.

Well, I think is time to make pcb-tools active again. Will start merging some MR and getting the house back on track.

I developed pcb-tools with lots of help from the community and was really useful for me, really glad it still makes sense and is useful for other people.

Let's get it moving again!

Give me some days and we'll get some momentum back.

xxxajk, 2023-04-10:
Great! Because I really lack time to adopt yet another abandoned repo that's incredibly useful.