curtacircuitos/pcb-tools #228: [IMPORTANT] is it time to designate https://github.com/xxxajk/pcb-tools as alive fork?

rusefillc, 2023-03-22:
https://github.com/xxxajk/pcb-tools has some commits

this repo seems to be abandoned https://github.com/curtacircuitos/pcb-tools/issues/227?

xxxajk, 2023-04-09:
Great... So you want me to adopt another project? :-)
I'm not sure if I have actual time for it, but I could hand out abilities for some extra people who are willing to help...

Argmaster, 2024-06-16:
Hi! For those seeking actively maintained alternative, let me introduce you to [PyGerber](https://github.com/Argmaster/pygerber). Although I'm sorry for posting it here, I think that after 3 years of no activity this should be either a wake up call for maintainer or convinience for community.

phsilva, 2024-06-19:
Hi @Argmaster , don't be sorry for posting. PyGerber looks fantastic. I will add a note on README marking this is unmantained and pointing interested people to your project. Thanks for the tip!